davesrocketshop/Rocket #11: Icons not appearing in wb dropdown or tool icons

concretedog, 2021-02-12:
Hi Dave, Excellent work. On my install all the icons are missing. Both the icon on the dropdown workbench and the actual tool icons. 

`10:22:47  Cannot find icon: /home/concretedog/.FreeCAD/Mod/Rocket/Resources/icons/RocketWorkbench.svg
10:23:04  Cannot find icon: /home/concretedog/.FreeCAD/Mod/Rocket/Resources/icons/Rocket_NoseCone.svg
10:23:04  Cannot find icon: /home/concretedog/.FreeCAD/Mod/Rocket/Resources/icons/Rocket_Transition.svg
10:23:04  Cannot find icon: /home/concretedog/.FreeCAD/Mod/Rocket/Resources/icons/Rocket_BodyTube.svg
10:23:04  Cannot find icon: /home/concretedog/.FreeCAD/Mod/Rocket/Resources/icons/Rocket_CenteringRing.svg
10:23:04  Cannot find icon: /home/concretedog/.FreeCAD/Mod/Rocket/Resources/icons/Rocket_Bulkhead.svg
10:23:04  Cannot find icon: /home/concretedog/.FreeCAD/Mod/Rocket/Resources/icons/Rocket_Fin.svg
`

davesrocketshop, 2021-02-12:
The files are in the release, but if you did a straight unzip, they may be located in /home/concretedog/.FreeCAD/Mod/Rocket-1.0 instead of /home/concretedog/.FreeCAD/Mod/Rocket. If that is the case, rename the folder. If it is not, please grab a screenshot of your /home/concretedog/.FreeCAD/Mod/Rocket folder.

concretedog, 2021-02-12:
Yes, that has solved it, I had DL'd and unzipped manually,

davesrocketshop, 2021-02-12:
Good. I'll update the manual install instructions.

closed 2021-02-12 by davesrocketshop