davesrocketshop/Rocket #22: Please add launch rail guides and buttons

livetoski78, 2022-01-08:
I would love to see the ability to make radiused launch rail guides and rail buttons for 1010 and 1515 rails.  A way to secure them through the airframe would be a huge plus!

davesrocketshop, 2022-01-08:
This has already been implemented in a dev branch, with a lot of options including airfoiled versions. Attaching to the rocket is a whole different problem that I haven't resolved yet.

TL;DR coming soon.

davesrocketshop, 2022-01-18:
You can preview the rail button design in the 'rocket_editor' branch. This is a development branch so there are many new features not working, but there are also a lot of new components in there that are working fine. Launch lugs, rail buttons, and rail guides are all new.

[image]

davesrocketshop, 2022-03-22:
Included in release 2.3.0

closed 2022-03-22 by davesrocketshop