davesrocketshop/Rocket #29: The width of the number input fields is too short.

BlindDuck, 2022-05-16:
The width of the number input fields is too short.
The green "checked icon", the unit and the values after the ./,## are displayed only, but not the leading ones.

[code]
OS: openSUSE Leap 15.3 (KDE//usr/share/xsessions/kde-plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28861 (Git)
Build type: Release
Branch: master
Hash: 4fea37b3bc5a3a9fce789dac6d2444f240bd3313
Python 3.6.15, Qt 5.12.7, Coin 3.1.3, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods:
  * Rocket 2.4.1
  * ThreadProfile 1.81.0
  * A2plus 0.4.56
  * Assembly4 0.11.12
[/code]

[image]

davesrocketshop, 2022-05-17:
I'm not sure what's happening here, and it's not something I'm able to reproduce. The fields in your display do not appear to be following my size specifications, which is a fixed size field 100px in size. I can change that to a minimum size spec, and probably will, but that still doesn't explain what you're seeing and may not fix it.

It appears you have a preference pack applied. I'm not sure if that would cause the problems or not. I would suggest to revert to the default UI settings and seeing if that solves the problem. If it does, we can work from there.

BlindDuck, 2022-05-17:
I only comieled from git and did not change any setting right now.
Please verify with your settings to pin point it further.
[image]

davesrocketshop, 2022-05-17:
I'm sorry, but I'm just not able to reproduce it. I've tried matching the settings to yours, including the language in case that had some significance.

I'm noticing some discrepancies between your settings and what is being displayed. For example, the 'No stylesheet' setting should normally produce the light colour backgrounds, not the dark colour. Your icons also seem smaller, but that may be difference in screen size. I've used the latest daily build, but you might have some differences having built from source.

It may well be screen size. Mine maxes out at 2560x1440, so I can't change to match yours. Can you try yours at the lower resolution?

If that turns out to be the problem, it should be fixed by changing the width from a fixed width to a minimum width.

BlindDuck, 2022-05-27:
I took some screen shoots (Windows/Linux, Part Design/Rocket) on a HP X2 1013 G3 (3000x2000).
The Part Design fields look the same but not the Rocket ones.
With lower resolution it got worse, only the green checked icon is visible.

[image]
[image]
[image]
[image]
.

davesrocketshop, 2022-07-15:
Merged

closed 2022-07-15 by davesrocketshop