davesrocketshop/Rocket #30: blunted ogive cone deformed

Arogen, 2022-05-24:
I create a nose cone with the following parameters:
blunted ogive (no shoulder)
hollow
272.61 length
66.04 diameter
1.50 thick
20.68 blunted diameter

The cone as it appears in FreeCad appears correct, if it export it to STL then import the STL to Tinkercad the footprint of the cone is 66.04 x 66.03, and there the cone is slightly deformed near the bottom.  I see the same deformity loading the cone into Cura.  Cones made using other parameters appear correct.  See the attached image
[image]
.

davesrocketshop, 2022-05-24:
Short answer is that the workbench has no control over the STL generation. That's done by FreeCAD.

The good news is that the defaults aren't well suited to nose cones, or curved surfaces in general. Please read this thread for a discussion: https://github.com/davesrocketshop/Rocket/issues/25

Try playing with the defaults. If that doesn't fix the issue let me know.

Arogen, 2022-05-26:
Thanks, that's what I needed.

Arogen, 2022-05-26:
> I think I need to do a video on this. I know you're not the only one who experiences this, and the fix while simple is not obvious.
>
> Glad it worked for you.
>
>
> This email has been checked for viruses by AVG.
> https://www.avg.com

A video would be good.  Point out that by changing those settings the file size increases (I assume the triangle count increases too).  I was planning to do some post processing in Tinkercad but couldn't due to its size/triangle limits.

 

Paul