davesrocketshop/Rocket #60: Tooltips for selected tools

kaktusus, 2023-08-01:
`version of the workbench 3.2.0

All tools except these two have pop-up tooltips.

Rocket Stage
[image]

Rocket Parallel Stage
[image]
`
cool example
[image]

davesrocketshop, 2023-08-02:
This seems to be a 0.21 issue. It works fine in 0.20. Needs more investigation.

kaktusus, 2023-08-02:
So I will also provide the FreeCAD version in each submission in the future.

hasecilu, 2024-06-10:
Is it still an issue? Working w/wo translations.

hasecilu, 2024-11-04:
Tooltip show fine, probably problem was because tooltip was the same as the menutext. BUt right now there are no translations on the WB, no `qm` files.

closed 2025-10-31 by davesrocketshop