davesrocketshop/Rocket #62: [GUI translation] →  main menu / Rocket

kaktusus, 2023-08-02:
```version of the workbench - 3.2.0```

The menu label and some of the submenus do not have translators.

[image]

1. `Rocket` - string is in Crowdin
    https://crowdin.com/translate/freecad-addons/32156/en-pl#6588168

2. ` Calculators` - string is in Crowdin
    https://crowdin.com/translate/freecad-addons/32156/en-pl#6588170

3. `Analysis` - string is in Crowdin
    https://crowdin.com/translate/freecad-addons/32156/en-pl#6600262

davesrocketshop, 2023-08-02:
The translation code is in there, but the wrong translation function was being used.

davesrocketshop, 2023-08-02:
[image]

closed 2023-08-02 by davesrocketshop