davesrocketshop/Rocket #63: Work bench fails to launch

Jsevenh, 2023-10-13:
rocket work bench version 3.2.1 master branch
[image]
I'm getting the above error while loading the workbench on freecad version 0.21.1
This is the report view output
// report view 
09:26:21  No module named 'PySide2.QtCharts'
09:26:21  Traceback (most recent call last):
  File "<"string">", line 49, in Initialize
  File "$HOME/snap/freecad/common/Mod/Rocket/./RocketGui.py", line 59, in <module>
    from Ui.Commands.CmdFlutterAnalysis import CmdFinFlutter
  File "$HOME/snap/freecad/common/Mod/Rocket/./Ui/Commands/CmdFlutterAnalysis.py", line 34, in <module>
    from Ui.DialogFinFlutter import DialogFinFlutter
  File "$HOME/snap/freecad/common/Mod/Rocket/./Ui/DialogFinFlutter.py", line 35, in <module>
    from PySide2.QtCharts import QtCharts

davesrocketshop, 2023-10-13:
Due to licensing issues, the dependency is not automatically included in the Linux versions of FreeCAD. I'll be removing this dependency in a future release.

In the meantime you can manually add it to your system. On ubuntu and likely most Debian derived systems you can do this with:

 $ sudo apt install python3-pyqt5.qtchart

Jsevenh, 2023-10-13:
Thank you for your response. In addition to your suggestion of using 'sudo apt install python3-pyqt5.qtchart,' I've also attempted to resolve the issue by running 'pip install PySide2.' Unfortunately, I'm still encountering the same error.

davesrocketshop, 2023-10-13:
Well, let's start from the beginning. From the FreeCAD menu bar, select "Help"->"About FreeCAD". The dialog will show some information, including a button saying "Copy to clipboard". Push the button, and then paste here

Jsevenh, 2023-10-14:
> Well, let's start from the beginning. From the FreeCAD menu bar, select "Help"->"About FreeCAD". The dialog will show some information, including a button saying "Copy to clipboard". Push the button, and then paste here

[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git) Snap 759
Build type: Release
Branch: tag: 0.21.1
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Rocket 3.2.1
  * A2plus 0.4.60n
  * Beltrami 1.0.8-alpha
  * CfdOF 1.24.5
  * lattice2 1.0.0
  * fasteners 0.4.66
  * Plot 2022.4.17
  * Curves 0.6.13
[/code]

davesrocketshop, 2023-10-14:
Ok, problem identified.

You're running FreeCAD from a snap. The key feature of the snap is that the program is packaged with its own dependencies, meaning it's not reliant on the packages installed by the OS. The other side of that coin is that installing missing packages on your OS will not be seen by the snap. So we're all out of luck there.

Plan B is to remove flutter analysis, and that will involve editing some lines of code. in the file `~/.local/share/FreeCAD/Mod/Rocket/Ui/Commands/CmdFlutterAnalysis.py` you'll have to change lines 34-48 to look like this:

```
# from Ui.DialogFinFlutter import DialogFinFlutter
from Ui.Commands.Command import Command

 def calcFinFlutter():

    # See if we have a fin selected
    for fin in FreeCADGui.Selection.getSelection():
        if fin.isDerivedFrom('Part::FeaturePython'):
            if hasattr(fin,"FinType"):
                # try:
                #     form = DialogFinFlutter(fin)
                #     form.exec_()
                # except TypeError as ex:
                #     QtGui.QMessageBox.information(None, "", str(ex))
                return
```

Unfortunately it's a bit of a project to replace this code and it's going to be a while before I can get to it.

Jsevenh, 2023-10-14:
Thanks , I really appreciate the help .

Jsevenh, 2023-10-14:
While this may not pertain to the current topic, I'd like to inquire about the most effective approach for initiating workbench development.

davesrocketshop, 2023-10-14:
I've just created a new release that handles the missing QtCharts module more gracefully. You still won't be able to use the flutter analysis dialog without it, but all other functionality will be unaffected.

davesrocketshop, 2023-10-14:
> While this may not pertain to the current topic, I'd like to inquire about the most effective approach for initiating workbench development.

Start here:
https://wiki.freecad.org/Workbench_creation

Good ways to communicate more interactively are through twitter @davesrocketshop ro via the FreeCAD discord channel

davesrocketshop, 2023-10-14:
I'm going to leave this issue open. The last release allows for graceful failure, but it is still failure not a fix.

The fix is to remove the requirement for QtCharts

davesrocketshop, 2023-10-15:
Fixed in v3.2.3.

The dependence on QtCharts has been removed, and the flutter dialog uses matplotlib instead. This package is part of the FreeCAD install and should be available on all supported platforms.

You should have full functionality with no need for any work arounds.

closed 2023-10-15 by davesrocketshop

Jsevenh, 2023-10-15:
Your efforts are greatly appreciated