davesrocketshop/Rocket #65: Changing placement of fins causes them to jump to strange places after copying a fin object

livetoski78, 2023-11-25:
Expected: After making multiple copies of a fin object, I should be able to rotate them and move them around.

Actual: Rotating or moving a copied fin causes the fin objects to be moved to strange places.

Steps to reproduce:
1.  Create a body tube
2.  Create a single fin
3.  Place it where you want it to be on a body tube
4.  Duplicate or copy/paste the fin
5.  Go to placement and select the "X" axis and "Apply incremental changes"
6.  Click "Ok" and notice the fin jumps position
[image]
[image]
[image]

davesrocketshop, 2023-11-25:
Not surprised by this which is probably not what you expected to hear. Let me explain why.

The fin is offset from the origin by the radius of the body tube. Even when not using the rocket builder there's an offset (this probably shouldn't be the case. I'll have to check to see what would be involved with a fix) Rotating a fin is really rotating it around the base + offset, In your context that offset is undefined. Even in the context of the rocket builder which is aware of the body tube diameter, rotation is not straight forward.

What "should" happen, which is also not what you want, is the fin rotation angle is the rotation angle from the base. You would still need to calculate an appropriate X and Y offset.

Let me look at appropriate solutions, but it won't be using the placement UI as you do in the second image, but setting appropriate property values as in the third image, or through the component UI. You would set the rotation angle there

livetoski78, 2023-11-25:
Check back a version or two, as this used to work flawlessly using that workflow.  It was only introduced a couple of months ago…I want to say two updates back, but may be mistaken.  I think it was the update that introduced fin cans.-Eric====================Sent from my iPhone

davesrocketshop, 2023-11-25:
That makes sense. Let me look into it. It may not be a simple reversion as there were a lot of changes to support the rocket builder.

davesrocketshop, 2025-08-05:
Working in current dev tree

closed 2025-08-05 by davesrocketshop