davesrocketshop/Rocket #66: Feature Request:  Ability to add threads to round objects such as body tubes, fin cans, etc.

livetoski78, 2023-11-25:
This would greatly ease trying to make threaded rocket parts, as typical thread creation in FreeCAD is quite an involved process (and one I haven't figured out yet)

davesrocketshop, 2023-11-25:
Interesting application. The hard part will be generalizing the UI

livetoski78, 2023-11-25:
Indeed.  From what I’ve compiled so far from the documentation, it involves sketching a triangle shape, helixes, and sweeping objects, then some cutting and pasting the ends on.  I know there are guys out there doing it, but it’s way beyond my rudimentary CAD skills.  Ideally, I want to create a screw in electronics bay in the nose cone for my featherweight trackers.  Right now, it’s a hack job using a printed plate and screws.  Also, screw together body tubes would be nice to break down larger rockets for transport and easier printing (I can do up to 12” length, currently).-Eric====================Sent from my iPhone