davesrocketshop/Rocket #69: Add optional dependencies for cfdof workench to package.xml based on the upcoming major feature anounced regarding CFD studies

luzpaz, 2024-10-11:
Per https://x.com/DavesRocketShop/status/1844743922947428491

When the time is right, a reminder that it would be helpful to add something like the following to the package.xml file:  

```xml
<!-- Optional install of 3rd-party CfdOF workbench if utilizing CFD studies -->
<depend type="addon" optional="true">CfdOF Workbench</depend>
```

Giving the user a heads-up about this functionality. Apologies if I'm being to presumptuous and you knew this already.

davesrocketshop, 2024-10-11:
Thanks. I'd already been thinking of this, but hadn't researched the optional tag yet. You saved me some digging :)

luzpaz, 2024-10-11:
Great, that was the scenario  that I was hoping to address!

closed 2025-08-05 by davesrocketshop