davesrocketshop/Rocket #73: Unknown command 'Rocket_CFDAnalysis' but CFDof already installed

FroppyKeropy, 2025-07-18:
[image]

Installed CFDof correctly, i already sucessfully set up simulations on other models so  i think its working, maybe additional setup is required? but already read the readme and it doesnt specify how to set up CFDof to enable 'Rocket_CFDAnalysis' commands on the rocket workbench 

all help is aprecciated

davesrocketshop, 2025-07-18:
There are some other python dependencies that don't always get installed when upgrading the Rocket WB. It's possible these are missing. The easiest way to ensure you have everything is to uninstall the Rocket WB, restart FC, and then reinstall.

FroppyKeropy, 2025-07-18:
Worked! thank you!

closed 2025-07-18 by FroppyKeropy